mccloudaero/evpr #1: New Integrated Electronics

petermccloud, 2018-09-19:
Electronics currently consist of an [Adafruit Huzzah32](https://www.adafruit.com/product/3405) and a custom power distribution board that takes the 7.4V from the (2) 3.7V LiPo batteries and provides power to the servos and the Huzzah32.

Need to design a custom integrated board to replace everything. Min requirements are:

1. ESP32 module
2. USB Port for charging and flashing
3. Power handling and charging for the 2 LiPos

Flynnn, 2019-06-16:
Design Requirements:
- [ ] Accept input power from ~7.4V to 16V
- [ ] Convert to TBD internal power voltage
- [ ] When powered: Utilize internal power level to charge batteries, drive Servos, drive Control
- [ ] When unpowered: Utilize battery power to drive Servos, drive Control
- [ ] Transition from powered to unpowered should not cause power interruption for servos or control
- [ ] Self-Shutdown line from WROOM (kill power until supply reset)
- [ ] External connections must be vibration tolerant.
- [ ] Support tachometer signal at power voltage
- [ ] Servo operating range: 5V to 9V (ideally 6V to 7.4V)
- [ ] Function as motherboard for WROOM esp32, especially USB to UART function
- [ ] Should be able to power and flash WROOM esp32 via USB without powering the rest of the circuit (Except LEDs)
- [ ] Support LED control via DotStar protocol (SPI)
- [ ] Batteries are two LiPos, total voltage 7.4V nominal
- [ ] Onboard status LED
- [ ] Flash LEDs as warning when board is plugged in to USB

Flynnn, 2019-06-16:
Current requirements:
2 Servos: Hi-Tec HSB-9475-SH
For each servo: 900mA at 7.4V max. 
Wroom: Less than 1100mA max at less than 4V.
DotStar one meter, 2000mA max at 5V.
Misc power budget: 1000mA max at 5V.

Total Power consumption: 32720 mAV (32.7 watts)
Total Power consumption, servos 4 A: 78600 mAV (78.6 watts)

Flynnn, 2019-06-16:
Appropriate power unit:
https://www.mouser.com/ProductDetail/ABB-Embedded-Power/APXW012A0X3-SRZ?qs=sGAEpiMZZMsc0tfZmXiUnQPUSHORMpN3VSVfnexIQAphp%2F3aJNtoDg%3D%3D

APXW012A0X3-SRZ

Flynnn, 2019-06-16:
Appropriate Battery Charge Circuit:
BQ2057WSNTR 
https://www.mouser.com/ProductDetail/Texas-Instruments/BQ2057WSNTR?qs=sGAEpiMZZMsfD%252BbMpEGFJW1dVgmxdCLczZWdZJsP7c0%3D

Flynnn, 2019-06-16:
Connectors exceeding 3A should be JST-VH:
www.jst-mfg.com/product/pdf/eng/eVH.pdf

Flynnn, 2019-06-16:
JST-VH board mount vertical 3-pin:
https://www.digikey.com/product-detail/en/jst-sales-america-inc/B3P-VH-LF-SN/455-1640-ND/926548

Flynnn, 2019-06-16:
Main-Line Fuse:
https://www.digikey.com/product-detail/en/littelfuse-inc/0501010.WR/F1625CT-ND/2620103

Flynnn, 2019-06-17:
Aha!

AOD417
https://www.digikey.com/products/en?keywords=AOD417

Has a v_gs (th) between 1V and 3V, which means we can safely allow the supply voltage to be up to 0.9V less than the charge voltage. So if charge is 8.2, we can safely power the system on 7.3V

Flynnn, 2019-06-17:
In accordance with the above discovery, and moving forward with APXW012A0X3-SRDZ, I am speccing servo voltage at 7.5V. This will give us a minimum 0.1V safety margin if the charge voltage is set to 8.4V

Flynnn, 2019-06-18:
Recommend 9.95kohm trim resistor. Technically, this produces min junction voltage of 7.28V, which is slightly less than the absolute minimum of 7.3, but that minimum was designed to give 100mV margin from the *true* minimum which is 7.2. Speccing to 9.95kohm will yield 85mV margin in the absolute worst case.

Flynnn, 2019-06-18:
Charge rate will be 0.5A. Minimum battery voltage is 6V. Maximum input power is ~15V. Maximum Vdrop is 9V. Max power dissipation must be 150% 4.5W = 6.75W.

Charge controller CC current target recommended at 35mA. Beta must be at least  0.5A/35mA = 15. (IE, the only BJTs in existence that would fail this margin are the leftovers from when BJTs were first invented).

Flynnn, 2019-06-18:
STD2805

https://www.mouser.com/ProductDetail/STMicroelectronics/STD2805T4?qs=%2Fha2pyFadujTwifOThaVPcZ6uCClVMZvJr7GIDYv1u0%3D

Seems appropriate

Flynnn, 2019-06-26:
For BQ2057T (or even BQ2057W) V_SNS is 125mv. Current sense resistor is calculated as 125mv/(0.5A) = 0.25 ohms.

Flynnn, 2019-06-26:
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/CSR0603FKR250/CSR0603FKR250TR-ND/1742486 Looks like an appropriate sense resistor

Flynnn, 2019-06-26:
Run of the mill 1k resistors are good for the CC gate resistor and the TS voltage divider

Flynnn, 2019-06-26:
Power system is essentially done with the following exceptions:

Input power safety needs to be completed (TVS addition, and fuse re-spec).

Self-Shutdown needs to be implemented. Tie a series enhancement PMOS gate to raw input through large resistor, add low power SCR to ground? Once triggered, SCR will remain active and force the PMOS off. Need to spice it out to confirm.

Connector for servos needs to be added. Filter caps for servos need to be added.

Once these are done, all that will remain will be ESP3-related things.

petermccloud, 2019-07-22:
Identified brushes to be used to transfer vehicle power to the EVPR rotors.
https://www.helwigcarbon.com/shop/99-362361/
https://www.helwigcarbon.com/shop/11-313761/
Both are a copper graphite brush, grade K136 from Hellwig Carbon and have a very low voltage drop (<0.7 V). Capacity for 99-362361 is 11 amps, while the capacity for 11-313761 is 16 amps. Moving forward with the 11-313761 for now to ensure extra margin.

Flynnn, 2019-07-22:
I am second-guessing the inclusion of a TVS. I did find one here:
https://www.digikey.com/product-detail/en/littelfuse-inc/SP4023-01FTG-C/F6757CT-ND/5428761

But here is the problem:
These guys are basically a couple zeners which temporarily short the power supply and ground so that if input voltage goes past a certain threshold, the short causes a vdrop bringing the short back down.

The problem is that this shorting is high-current. If we put it after our fuse, it will basically cause the fuse to blow whenever there are transient voltage spikes, which we absolutely do not want to happen in flight.

We can't use the TVS without a fuse, as, were it to fail (rather than trip), it would create a permanent dead short.

Using the TVS with its own separate fuse is an option, but is getting on the edge of ridiculousness.

At the end of the day, I think having some faith in the output of the inverter, and using some good decoupling capacitors will be all that is necessary to prevent transient voltage spikes.

Flynnn, 2019-07-22:
1W, 2K resistors for latching power off:
https://www.digikey.com/product-detail/en/bourns-inc/CHV2512-FX-2204ELF/CHV2512-FX-2204ELFCT-ND/5176019

Flynnn, 2019-07-22:
Forget SCR! Can use a P-MOSFET flip flop instead.

I'm gonna use a third power FET for now, as I don't think the increase in cost or board space will be enough to cause any problems, and the re-use of components simplifies the design.

Flynnn, 2019-07-23:
Forget a third FET! While a flip-flop is possible, it will require 12V control logic. That can be achieved using a BJT, but then, why not just use the BJT directly? I will spec a BJT to be used for a latching shutdown.

Flynnn, 2019-07-23:
For the NPN controlling the shutdown latch:
https://www.digikey.com/product-detail/en/nexperia-usa-inc/PMBT4401235/1727-1886-1-ND/5015561

Flynnn, 2019-08-02:
A far more elegant solution to latched power down is to use an SR latch. This will require PoR circuit.

6.85V is the minimum voltage our power switcher can produce (charging voltage minus the FET threshold voltage, minus the schottky forward voltage). Therefore, a safe voltage to set as the power-off threshold is 5V.

For simplicity, the latch should be rated for up to 16V.

Recommend using R3119N050A PoR chip:
https://www.mouser.com/ProductDetail/Ricoh-Electronic-Devices-Company/R3119N050A-TR-FE?qs=sGAEpiMZZMunEhqKs81nFAtn9C%252BfdOcC%2F8Cteh6vBNc%3D

Device accepts external delay capacitor, and will output LOW for limited period of time after power on.

That leaves just the SR latch itself. There is very little room for creativity, here -- there are three manufacturers making parts in our regime, and they all appear to be essentially identical. (In fact, they are all either 4044B, or 4043B series). The main thing is we want one that is active LOW, as this will enable us to avoid using an inverter, meaning we want a 4044B part. Recommending:

https://www.digikey.com/product-detail/en/on-semiconductor/MC14044BDR2G/MC14044BDR2GOSCT-ND/3462303

Flynnn, 2019-08-02:
Unfortunately, SR latches at this voltage will not respond to the logic levels of the ESP32. Therefore, an intermediate transistor is still required.

One of the BJTs I specced earlier will suffice.

Flynnn, 2019-08-02:
Updated Design Requirements to reflect the latest plan for USB power/flash support.

One specific problem: Our 5V DotSTAR array can, in theory, draw more current than is available on the USB line. This means that, in theory, it is possible for a software failure to lead to a current overdraw, and potentially damage any computer that is plugged in. This is unacceptable. To fix this, a 500mA fast-blow fuse should be in-line with the power rail for the USB. This will render USB-power-on damaged in the event of a software failure, rather than the programming computer, and will allow communications to remain functional if the main board is powered from battery.

Flynnn, 2019-08-02:
Recommend the following as USB Fuse:
https://www.digikey.com/product-detail/en/bel-fuse-inc/0686F0500-01/507-2156-1-ND/7427708

Flynnn, 2019-08-02:
Utilizing this USB port:
https://www.digikey.com/product-detail/en/amphenol-icc-fci/10118193-0001LF/609-4616-1-ND/2785380

Flynnn, 2019-08-02:
Recommend the following Schottky for 5V source combination:
https://www.digikey.com/product-detail/en/microsemi-corporation/LSM115JE3-TR13/LSM115JE3-TR13CT-ND/1634547

Flynnn, 2019-08-02:
Recommend the following for 5V switching regulator:
https://www.mouser.com/ProductDetail/CUI/V7805-2000R?qs=sGAEpiMZZMsc0tfZmXiUnYsiwROEjoqDRrA2Q91xKwgtSe9VhRcFkg%3D%3D

Flynnn, 2019-08-03:
Recommend switching to the latest USBXpress UART interface:
https://www.digikey.com/product-detail/en/silicon-labs/CP2102N-A02-GQFN24R/336-5888-1-ND/9863482

Flynnn, 2019-08-03:
Note To Self: https://www.random-science-tools.com/electronics/SMD_markings.php?formtype=SMD_markings_search&search_string=AC1A is an amazing tool.

Flynnn, 2019-08-04:
List of VSENSE/STAT arrangements:
VBUS & USB Voltage Sense: 47.5kohm:22.1Kohm
Raw Power Sense: 1:7
Battery Power Sense: 1:2
Charge STAT Sense: Direct, but use resistor on challenge pin

Flynnn, 2019-08-04:
Recommend these as LEDs:
https://www.digikey.com/product-detail/en/lite-on-inc/LTST-C171GKT/160-1423-1-ND/386792

Flynnn, 2019-08-04:
LEDs have max current 30mA. Spec to 20mA.

At this current, forward voltage is 2.2V.

500ohms will suffice for both the 12V to 16V input, and our 9V indicator.
250 ohms will suffice for the 5V indicator.

Flynnn, 2019-08-04:
Therefore, in total, we need to spec the following additional resistor types:

500 ohm, 0.5W resistor:
250 ohm, 0.5W resistor:
22.1Kohm resistor:
47.5kohm resistor:

petermccloud, 2019-08-06:
Changed bolt spacers to 6061-T6 and reduced diameter to 3/8". Updated footprint to reflect.
https://github.com/mccloudaero/evpr/blob/alt_gearing/electronics/available_foot_print_in.dxf

petermccloud, 2019-08-25:
Dimensions for brush plate power paths
https://github.com/mccloudaero/evpr/blob/combined_board/electronics/brushes/brush_plate_power.pdf

petermccloud, 2019-08-25:
PCB will be mounted via #6 screws that thread directly into the plates. Stainless steel washers will be used to distance the plate away from the plate. 1/32" nominal thickness, ID .143", OD .267
https://www.mcmaster.com/90945a716

petermccloud, 2019-08-25:
Mounting screws will be #6-32, stainless steel with 3/16" thread lengths. Head diameter is .262".
https://www.mcmaster.com/92949a143

petermccloud, 2019-08-25:
Added recommended screw hole locations around the interior of the board:
https://github.com/mccloudaero/evpr/blob/combined_board/electronics/available_foot_print_in.dxf

Flynnn, 2019-09-09:
Reset Button:
https://www.digikey.com/product-detail/en/c-k/PTS645SK43SMTR92-LFS/CKN9084CT-ND/1146821

Flynnn, 2019-09-09:
3.3V regulator 
https://www.digikey.com/product-detail/en/diodes-incorporated/AP2112K-3.3TRG1/AP2112K-3.3TRG1DICT-ND/4505257

Flynnn, 2019-09-09:
DotStar Level Shifter
https://www.digikey.com/product-detail/en/texas-instruments/SN74AHCT125DR/296-4652-1-ND/405208

Flynnn, 2019-09-09:
@petermccloud One meter of DotStars is limited to 2 amps. I have just realized that the power merging Schottky diodes I selected are 1 amp maximum.

Should I select new schottky diodes, or should I just let the official power limit be 1 amp?

Also, we should add a fuse to the DotStars so that an accidental over-current blows that fuse, rather than taking down the entire unit.

I will select a 1 amp fast blow fuse.

Flynnn, 2019-09-09:
DotStar Protection Fuse:
https://www.digikey.com/product-detail/en/bel-fuse-inc/C1Q-1/507-1077-1-ND/615050

Flynnn, 2019-09-09:
DotStar Connector: https://www.digikey.com/products/en?keywords=b4p-vh(LF)(SN)

Flynnn, 2019-09-09:
IO13 (WROOM pin 20, physical 16) and IO27 (pin 16, physical 12)

Flynnn, 2019-09-09:
Okay, this is a conundrum. I have no idea why on earth the DotStar data and clock pins are wired to where they are wired on adafruit boards. I am going to just follow suit as a stopgap, but we should investigate this at some point to understand what is going on

Flynnn, 2019-09-09:
PAD connector:
https://www.digikey.com/product-detail/en/harwin-inc/M20-9960245/952-2244-ND/3728208

Flynnn, 2019-09-09:
Servo Connector: https://www.digikey.com/products/en?keywords=B3P-VH(LF)(SN)

Flynnn, 2019-09-10:
Servo 1200mA * 2 * 8.4V
Wroom: 1100mA * 5.0V
DotStar: 1000mA * 5V
Misc: 1000mA * 5V

Total estimated consumption: 35660mAV
With factor of safety: 53490mAV
Total current at 12V: 4.5A

Flynnn, 2019-09-18:
Ideal Diode Power ORing controller:
https://www.digikey.com/product-detail/en/linear-technology-analog-devices/LTC4353CMS-PBF/LTC4353CMS-PBF-ND/3305799

Flynnn, 2019-09-18:
Using an ideal diode ORing controller, all we need to do to ensure battery is focused on charging when main power operational is select a DC-DC converter with somewhere around 0.1V or more higher output than the max charging voltage for the charge controller. This max charging voltage is 8.2V currently. 8.3V or higher is an acceptable output voltage, but do not exceed 9V, as this is the maximum we are comfortable operating the servos on.

Flynnn, 2019-09-19:
For future selves: It turns out we picked the ridiculously overpowered power supply because it is literally the smallest and least expensive thing on the market that can output more than 4.5A at 8.4 some-odd volts. There is, however, one supply that I missed the first several passes:

https://www.digikey.com/product-detail/en/texas-instruments/TPSM84824MOLR/296-48381-2-ND/8543598

It's about half the size of our currently selected supply, and fits the requirements much closer.

Flynnn, 2019-09-19:
Okay, yeah, I really think this is the best part on the market for us. There are some lower spec chips available, but they are from the same series, and so will have roughly the same cost and identical footprint. No point in using them. Any other parts that closer match our spec are far more expensive (All exceeding $25, and some exceeding $80). This is the best we can do, I am pretty sure. I will adjust the design to accomodate the new supply, although the benefits (other than footprint and weight) are questionable, considering its specs are almost identical to the one we had selected before.

Flynnn, 2019-09-19:
Okay, so, running through the design procedure for this new power unit:

# VOut Select
Target VOut is 8.4V. To set, we must have R_FBT = 10k and R_FBB = 6/(V_OUT - 0.6) * kOhm

In other words, R_FBB = 6/(8.4 - 0.6) kOhm = 769 ohms, or thereabouts.

Flynnn, 2019-10-20:
For selecting the ORing controller mosfets:
VDSS limit should exceed 8.4V. Exceed 10V for safety.
VGSS limit should exceed 14V as demanded by the chip
Must carry 4.5A * 12V / 8.2V = 6.6A comfortably

Flynnn, 2019-10-30:
Sense activation/deactivation transistor:
https://www.digikey.com/product-detail/en/diodes-incorporated/DMN5L06K-7/DMN5L06KDICT-ND/1287193

It's an enhancement mode NFET with a low enough driving voltage to be driven directly by by the feather. It can handle at least 16V between all nodes, and up to 50V directly across DS. It should be a very copy-pasta-ble transistor for turning the various sense voltage dividers on and off.

Flynnn, 2019-10-30:
So, the simplest solution I can work out here is to use NFETs for low-side switching of the sense dividers. The problem is, when the dividers are disabled, this will cause the sense pins to exceed 3.3V. We should use a 3.3V zener diode on each sense pin to protect from over-voltage.

Flynnn, 2019-10-30:
0603 USB shielding capacitor (4.7nf) : https://www.digikey.com/product-detail/en/yageo/CC0603CRNPO9BN4R7/311-1740-1-ND/5195642

0805 USB shielding resistor (1M) : https://www.digikey.com/product-detail/en/yageo/RC0805FR-071ML/311-1.00MCRCT-ND/730392

[21 more replies]