mccloudaero/evpr #14: Fab Process Improvements

Flynnn, 2020-03-10:
Purchase sufficient organizers
Create two organizers: One for top, one for bottom, parts that are both top and bottom are given a compartment in each.
We should buy metal stencils
Manufacture a board holder for stencils

Flynnn, 2020-03-22:
OK, so the board holder is pretty much off-limits as that will require EPL access which is impossible.

I stand by my advocation for metal stencils, though!

Flynnn, 2020-03-22:
Schematic is now complete, so I can generate a BOM and identify top/bottom part counts.

Flynnn, 2020-03-22:
There are about 47 unique part types. At maximum we will require 94 compartments, with 4 being large.

Recommend purchasing:
8 (x10) of: https://www.adafruit.com/product/431 ($31.6 total)
8 (x3)  of: https://www.adafruit.com/product/432 ($23.6 total)
8 (x2) of: https://www.adafruit.com/product/433 ($23.6 total)

This will result in 120 compartments. The extra overhead is to account for inaccuracies in my estimation on final part count and box size selections. It also allows for the addition of new parts later in the project if need be. The total for this purchase is $78.8 + shipping.

Flynnn, 2020-03-23:
Okay, with Adafruit out of business for the foreseeable future, let's just buy one of the 100pcs kits here:

https://www.amazon.com/20PCS-Electronic-Component-Laboratory-Storage/dp/B01NA9WOX1

It'll cost $32

I can't find anyone selling the small or medium sizes, but the tiny size should at least be useful for most of what we need to do.

petermccloud, 2020-03-23:
Power Supply for anodizing
https://www.caswellplating.com/electroplating-anodizing/power-supplies/cc-cv-rectifier-0-30v-30-amp.html

petermccloud, 2020-03-23:
DigiKey Boxes:
431- https://www.digikey.com/product-detail/en/adafruit-industries-llc/431/1528-1139-ND/5353599
432 - https://www.digikey.com/product-detail/en/adafruit-industries-llc/432/1528-1140-ND/5353600
433 - https://www.digikey.com/product-detail/en/adafruit-industries-llc/433/1528-1241-ND/5356726

Flynnn, 2020-03-23:
You're a legend; fire away on those DigikKey Boxes! @petermccloud 

I still need to look at the PS

Flynnn, 2020-03-23:
That supply looks great! If that's what you want to order, by all means.

petermccloud, 2020-03-23:
@Flynnn I know you're looking for a quality hot air station. I did find the Quick 861DW [for $265 with free shipping](https://www.tequipment.net/Quick/861DW/Desoldering-Equipment/Rework-Stations/#description). However, looking at other options, I trust Adafruit and they also sell the Quick 957DW+, which I can [find for $95](https://www.tequipment.net/Quick/957DW+/Desoldering-Equipment/Rework-Stations/). Adding a 10 pack of different nozzle size brings it to $130. It's not AS stellar reviews, but still decent. Considering it's half the price, I'd need to be convinced to get the more expensive one.

petermccloud, 2020-03-23:
Ok. After looking over my anodizing requirements in more detail, a 10 amp supply should be more than sufficient. I'm looking at the next bigger size of what you pointed out, but the newer style supply which has a finer amperage control, which I'll need for anodizing.
https://www.amazon.com/dp/B07DQR177D/ref=emc_b_5_t

petermccloud, 2020-03-23:
@Flynnn Got everything ordered. Decided to just go with the more expensive hot air station. Ended up getting this power supply https://smile.amazon.com/dp/B073TW8H2S/ref=emc_b_5_t. It's just the 10A version of what you pointed me to. It has less noise and better voltage regulation than the newer style.