mccloudaero/evpr #16: Better silkscreen is needed

Flynnn, 2020-03-10:
Add our names, add component names too

Flynnn, 2020-03-21:
@petermccloud Would you be willing to copy a line-art Mccloud Aero logo for the silkscreen? Something that can be tiny, like fit in a 10mm by 7mm area?

petermccloud, 2020-03-21:
@Flynnn By line-art you mean SVG? Looks like I can't attach files here. I'll email you the files.

Flynnn, 2020-03-23:
```py
import pandas as pd
import re

pd.set_option('display.max_rows', None)
pd.set_option('display.max_columns', None)
pd.set_option('display.width', None)
pd.set_option('display.max_colwidth', -1)

bomraw = pd.read_csv('bomraw.csv', sep=';')
bom = bomraw[['Part', 'Value', 'COMPONENT']]

groups = {};

for part in bom['Part']:
    prefix = re.search(r'[a-zA-Z\.]+', part).group(0);
    if (prefix not in groups):
        groups[prefix] = []
    groups[prefix].append(part)
    

for k,v in groups.items():
    parts = v
    parts.sort(key = lambda part: int(re.search(r'[0-9]+', part).group(0)))
    n = 0
    for part in parts:
        n += 1
        prefix = re.search(r'[a-zA-Z\.]+', part).group(0)
        print(f'NAME {part} {prefix}{n};')
```

Flynnn, 2020-03-23:
NAME C.B13 C.B1;
NAME C.B32 C.B2;
NAME C.D15 C.D1;
NAME C.D21 C.D2;
NAME C.EJ1 C.EJ1;
NAME C.ES2 C.ES1;
NAME C.ES3 C.ES2;
NAME C.F9 C.F1;
NAME C.F12 C.F2;
NAME C.F15 C.F3;
NAME C.F17 C.F4;
NAME C.F20 C.F5;
NAME C.F26 C.F6;
NAME C.F35 C.F7;
NAME C.F36 C.F8;
NAME C.G39 C.G1;
NAME C.J27 C.J1;
NAME C.J28 C.J2;
NAME C.L30 C.L1;
NAME C.L31 C.L2;
NAME C.M1 C.M1;
NAME C.M2 C.M2;
NAME C.M3 C.M3;
NAME C.M4 C.M4;
NAME C.M5 C.M5;
NAME C.M10 C.M6;
NAME C.M29 C.M7;
NAME C.M40 C.M8;
NAME C.P16 C.P1;
NAME C.P22 C.P2;
NAME C.P23 C.P3;
NAME C.P24 C.P4;
NAME C.S37 C.S1;
NAME C.S38 C.S2;
NAME C.Y6 C.Y1;
NAME C.Y7 C.Y2;
NAME C.Y8 C.Y3;
NAME C.Y33 C.Y4;
NAME C.Z11 C.Z1;
NAME C.Z14 C.Z2;
NAME C.Z18 C.Z3;
NAME C.Z19 C.Z4;
NAME C.Z25 C.Z5;
NAME C.Z34 C.Z6;
NAME D.F3 D.F1;
NAME D.J1 D.J1;
NAME F.D1 F.D1;
NAME L.F20 L.F1;
NAME L.P2 L.P1;
NAME L.Z19 L.Z1;
NAME Q.B3 Q.B1;
NAME Q.F6 Q.F1;
NAME Q.F7 Q.F2;
NAME Q.J16 Q.J1;
NAME Q.K4 Q.K1;
NAME Q.L5 Q.L1;
NAME Q.L9 Q.L2;
NAME Q.L11 Q.L3;
NAME Q.Y1 Q.Y1;
NAME Q.Y2 Q.Y2;
NAME Q.Z8 Q.Z1;
NAME Q.Z10 Q.Z2;
NAME Q.Z13 Q.Z3;
NAME R.B3 R.B1;
NAME R.B4 R.B2;
NAME R.B5 R.B3;
NAME R.B6 R.B4;
NAME R.CD4 R.CD1;
NAME R.CD5 R.CD2;
NAME R.ED3 R.ED1;
NAME R.ED4 R.ED2;
NAME R.ES1 R.ES1;
NAME R.ES2 R.ES2;
NAME R.F17 R.F1;
NAME R.F18 R.F2;
NAME R.F19 R.F3;
NAME R.F30 R.F4;
NAME R.G10 R.G1;
NAME R.J15 R.J1;
NAME R.J20 R.J2;
NAME R.J21 R.J3;
NAME R.J22 R.J4;
NAME R.J23 R.J5;
NAME R.K9 R.K1;
NAME R.K28 R.K2;
NAME R.L24 R.L1;
NAME R.L25 R.L2;
NAME R.L26 R.L3;
NAME R.L27 R.L4;
NAME R.L33 R.L5;
NAME R.L34 R.L6;
NAME R.M1 R.M1;
NAME R.M8 R.M2;
NAME R.M11 R.M3;
NAME R.M32 R.M4;
NAME R.P12 R.P1;
NAME R.P13 R.P2;
NAME R.P14 R.P3;
NAME R.P16 R.P4;
NAME R.Y7 R.Y1;
NAME SJ.K2 SJ.K1;
NAME T.P3 T.P1;
NAME T.P5 T.P2;
NAME TP.B6 TP.B1;
NAME TP.F9 TP.F1;
NAME TP.F10 TP.F2;
NAME TP.F11 TP.F3;
NAME TP.F18 TP.F4;
NAME TP.J2 TP.J1;
NAME TP.J3 TP.J2;
NAME TP.L4 TP.L1;
NAME TP.L8 TP.L2;
NAME TP.M5 TP.M1;
NAME TP.M7 TP.M2;
NAME TP.P13 TP.P1;
NAME TP.P14 TP.P2;
NAME TP.P15 TP.P3;
NAME TP.P16 TP.P4;
NAME TP.P19 TP.P5;
NAME TP.Y1 TP.Y1;
NAME TP.Z12 TP.Z1;
NAME TP.Z17 TP.Z2;
NAME U.B1 U.B1;
NAME U.D6 U.D1;
NAME U.F2 U.F1;
NAME U.G3 U.G1;
NAME U.P12 U.P1;
NAME U.Y8 U.Y1;
NAME U.Z17 U.Z1;
NAME U7 U1;
NAME X.D5 X.D1;
NAME X.F1 X.F1;
NAME X.G1 X.G1;
NAME X.J1 X.J1;
NAME X.K1 X.K1;
NAME X.L1 X.L1;
NAME X.L2 X.L2;
NAME X.S2 X.S1;
NAME X.S3 X.S2;
NAME X.S4 X.S3;

Flynnn, 2020-03-23:
All components now follow new naming scheme: 76222ff

Scheme is as follows:

A.BXXX

where A is the component type prefix (U, Q, L, D, R, C, X, TP, SJ), and B is the block prefix (describing the functional block of origin for the part), and XXX is a unique number within the particular block/type prefix, counting starting from 1

Flynnn, 2020-03-25:
[mccloud_aero_logo_tiny_fo_silkscreen_202003.dxf.zip](https://github.com/mccloudaero/evpr/files/4384275/mccloud_aero_logo_tiny_fo_silkscreen_202003.dxf.zip)

Flynnn, 2020-03-25:
[mccloud_aero_logo_tiny_fo_silkscreen_20200_mm.dxf.zip](https://github.com/mccloudaero/evpr/files/4384295/mccloud_aero_logo_tiny_fo_silkscreen_20200_mm.dxf.zip)

Flynnn, 2020-03-25:
[mccloud_aero_logo_tiny_fo_silkscreen_202003.svg.zip](https://github.com/mccloudaero/evpr/files/4384311/mccloud_aero_logo_tiny_fo_silkscreen_202003.svg.zip)

Flynnn, 2020-03-26:
[Market_Deco.ttf.zip](https://github.com/mccloudaero/evpr/files/4384323/Market_Deco.ttf.zip)

Flynnn, 2020-03-26:
Complete  f3e1695

closed 2020-03-26 by Flynnn