mccloudaero/evpr #24: EVPR MK2 Rev 2 Parts Selection

Flynnn, 2020-03-21:
This issue is a part selection notebook for the second revision of the MK2 EVPR board.

Flynnn, 2020-03-21:
4.7uF 20% 0603 capacitor: CL10A475MQ8NNNC
https://www.digikey.com/product-detail/en/samsung-electro-mechanics/CL10A475MQ8NNNC/1276-1907-1-ND/3889993

Flynnn, 2020-03-21:
0.1uF 5% 0603 capacitor: 06033C104JAT2A
https://www.digikey.com/product-detail/en/avx-corporation/06033C104JAT2A/478-3713-1-ND/1116411

Flynnn, 2020-03-21:
450mV/3A SOD-123H Rectifier Diode: SMD34LHE1-TP
https://www.mccsemi.com/pdf/Products/SMD34LHE1(SOD-123HE1)-A.pdf

Flynnn, 2020-03-21:
4.7uH (1.5A sat / 80 mOhm): SRR4818A-4R7Y
https://www.digikey.com/product-detail/en/bourns-inc/SRR4818A-4R7Y/SRR4818A-4R7YCT-ND/7556055

Flynnn, 2020-03-21:
3.9uH (1.9A sat / 80mOhm): SRR4818A-3R9Y
https://www.digikey.com/product-detail/en/bourns-inc/SRR4818A-3R9Y/SRR4818A-3R9YCT-ND/7556054

Flynnn, 2020-03-22:
56µF 6.3mm Electrolytic: APXG250ARA560MF61G  [OLD since last rev]
https://www.digikey.com/product-detail/en/united-chemi-con/APXG250ARA560MF61G/565-4303-1-ND/5824824

Flynnn, 2020-03-22:
37.4 kOhm 1% 0603: RC0603FR-0737K4L
https://www.digikey.com/product-detail/en/yageo/RC0603FR-0737K4L/311-37-4KHRCT-ND/730134

Flynnn, 2020-03-22:
768ohm 0603: RMCF0603FT768R
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/RMCF0603FT768R/RMCF0603FT768RCT-ND/6053173

Flynnn, 2020-03-22:
JST PA 4POS: BM04B-PASS-1-TFT(LF)(SN) 
https://www.digikey.com/product-detail/en/jst-sales-america-inc/BM04B-PASS-1-TFT-LF-SN/455-2639-1-ND/2773434

Flynnn, 2020-03-22:
JST PA also has sufficient current rating for our servos. (roughly twice what is required, in fact). So, for space savings, recommend switching servo connectors to JST PA

Flynnn, 2020-03-22:
By extension, this will work for the PAD as well! So only our batteries need to be the big brother, JST VH

Flynnn, 2020-03-22:
JST PA 3POS: BM03B-PASS-1-TFT(LF)(SN) 
https://www.digikey.com/product-detail/en/jst-sales-america-inc/BM03B-PASS-1-TFT-LF-SN/455-2638-1-ND/2773433

JST PA 2POS: BM02B-PASS-1-TFT(LF)(SN)
https://www.digikey.com/product-detail/en/jst-sales-america-inc/BM03B-PASS-1-TFT-LF-SN/455-2638-1-ND/2773433

Flynnn, 2020-03-22:
Just double checked power calculations. Yes, our upstream power sources (Batteries / brushes) need to withstand closer 4.5A and 7.25A respectively. The converter will need to produce 6.4A.

Flynnn, 2020-03-22:
22uF 0603 GRM188R61A106KE69D:
https://www.digikey.com/product-detail/en/murata-electronics/GRM188R61A106KE69D/490-10474-1-ND/5026392

Flynnn, 2020-03-22:
22uF 0603: CL10A226MO7JZNC
https://www.digikey.com/product-detail/en/samsung-electro-mechanics/CL10A226MO7JZNC/1276-7076-1-ND/7320718

Flynnn, 2020-03-22:
1kohm 0603 RMCF0603FT1K00:
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/RMCF0603FT1K00/RMCF0603FT1K00CT-ND/1942996

Flynnn, 2020-03-22:
1Mohm 0603: RMCF0603FG1M00
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/RMCF0603FG1M00/RMCF0603FG1M00CT-ND/4425038

Flynnn, 2020-03-22:
1Mohm 0603: RMCF0603FT100K
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/RMCF0603FT100K/RMCF0603FT100KCT-ND/1943118

Flynnn, 2020-03-22:
22.1kohm RMCF0603FT22K1:
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/RMCF0603FT22K1/RMCF0603FT22K1CT-ND/1943077

Flynnn, 2020-03-22:
95.3k 0603: RMCF0603FT95K3
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/RMCF0603FT95K3/RMCF0603FT95K3CT-ND/2418160

Flynnn, 2020-03-22:
10kohm 0.1% RG1608P-103-B-T5:
https://www.digikey.com/product-detail/en/susumu/RG1608P-103-B-T5/RG16P10-0KBCT-ND/1240835

Flynnn, 2020-03-22:
47.5kohm 0603: RMCF0603FT47K5
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/RMCF0603FT47K5/RMCF0603FT47K5CT-ND/1943100

Flynnn, 2020-03-22:
300ohm 0603: RMCF0603JT300R
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/RMCF0603JT300R/RMCF0603JT300RCT-ND/2418198

Flynnn, 2020-03-22:
10kohm 0603: RMCF0603FT10K0 
https://www.digikey.com/product-detail/en/stackpole-electronics-inc/RMCF0603FT10K0/RMCF0603FT10K0CT-ND/1943057

GeorgesOatesLarsen, 2020-07-31:
33uH 5040 1.3A 180mOhm: SRN5040TA-330M\
https://www.digikey.com/product-detail/en/bourns-inc/SRN5040TA-330M/SRN5040TA-330MCT-ND/9653909

GeorgesOatesLarsen, 2020-07-31:
0.47uF 0603 ceramic: C0603C474K4PACTU 
https://www.digikey.com/product-detail/en/kemet/C0603C474K4PACTU/399-7934-1-ND/3471657

GeorgesOatesLarsen, 2020-07-31:
1uF 0603 ceramic: CL10A105KP8NNNC
https://www.digikey.com/product-detail/en/samsung-electro-mechanics/CL10A105KP8NNNC/1276-1182-1-ND/3889268