mccloudaero/evpr #35: Select optimal 5.0V and 3.3V converter capacitors

Flynnn, 2020-03-22:
Looks like I recycled the filtering capacitors for 8.4V converter. Fine and dandy, but they are absolutely massive. Recommend running through actual AP6320x design process to select caps.

Flynnn, 2020-03-22:
Output capacitor should be 0603 ceramic 22uF. 
Input capacitor should be 0603 10uF ceramic having RMS rating exceeding 0.5A.

Flynnn, 2020-03-22:
Input capacitor:

22uF 0603 GRM188R61A106KE69D:
https://www.digikey.com/product-detail/en/murata-electronics/GRM188R61A106KE69D/490-10474-1-ND/5026392

Flynnn, 2020-03-22:
Output Capacitor:
22uF 0603: CL10A226MO7JZNC
https://www.digikey.com/product-detail/en/samsung-electro-mechanics/CL10A226MO7JZNC/1276-7076-1-ND/7320718

GeorgesOatesLarsen, 2021-08-18:
We had to change to GRM188R61E106MA73J, the old part  GRM188R61A106KE69D, is no longer sold at digikey. Please update online BOM to match