mccloudaero/evpr #100: Magnet for RPM sensing

petermccloud, 2021-09-29:
Need a magnet in the brush holder for RPM sensing.

@GeorgesOatesLarsen Below is the link for the magnet I'm considering for RPM sensing. It's the same width and depth as the brushes. (1/4"x1/4") so I should have minimal issue incorporating it. If there's any potential issues you see, let me know.

https://www.mcmaster.com/5848K91/

GeorgesOatesLarsen, 2021-09-29:
The only issue I see is that it might be possible to place the magnet in the wrong direction, since it is a cube, which would render the hall effect sensor useless -- A disc magnet might be better?

@petermccloud

petermccloud, 2021-09-29:
@GeorgesOatesLarsen I think you're right about the disc shape. I can fit two of the below magnets in the brush holder.
https://www.mcmaster.com/5862K166

GeorgesOatesLarsen, 2021-09-30:
I'm not sure whether that would work very well; The magnetic pole needs to face perpendicularly to the axis of the rotor shaft. The height of those magnets makes me think you might be planning to have them parallel to the shaft?

Actually, those magnets seem pretty tiny to me, what about:
https://www.mcmaster.com/5862K141/
?

Also, I am thinking since it would be ideal to have two magnets with flipped poles, we end up having to be careful about the magnet orientation anyways. Although with disc magnets, its just important that we get the magnets facing the opposite direction as each other, whereas with cube magnets, there are a large number of other orientations that are incorrect

@petermccloud