mccloudaero/evpr #101: Current brush board shows signs of catastrophic wear

GeorgesOatesLarsen, 2021-09-29:
I am getting extremely inconsistent system voltage coming through the brushes on the rotor prototype that you dropped off with me, @petermccloud 

Here is a picture of the tracks under a microscope:
[image]

That regular grid pattern is particularly concerning, as that is presumably the weave pattern of the fiberglass substrate. I think we are wearing the copper down to the substrate.

I wasn't able to get any visible sparking, but this problem is much more serious.

There is definitely some wear and tear on the in-house track you made, but it has not gotten anywhere near wearing down to the substrate like that. I don't see a fiberglass weave on that board.

This problem was creating some pretty serious power interruptions on engine-power. Enough for the engine power indicator to visibly change brightness, for the current output on my power supply to significantly change, and for an oscilloscope read I had on the track to go berserk (though I don't have any good pictures of this). We're dropping down as low as, like, 6V, at times. It's real bad.

Since we are moving away from using a brush-based tach, maybe we should just adhere a copper ring to some blank FR4? Or something similar, that would allow some raw copper to be used? Perhaps we can cut some copper sheet into a shape such that we can screw it down onto the existing brush board? Something that would let us use significantly more copper than we currently are.

petermccloud, 2021-09-29:
Wow! Thanks for digging into that!

That brush PCB has only been spun around by hand. The previous boards have been spun by the motor. I thought both the homemade brush path and the OSHPark one were both 2oz copper. I'm trying to confirm what I got from Injectorall, but the invoice doesn't list the weight. It perhaps could have been 4oz.

The easiest solution is to return to using the original copper brush path boards, but omit the tach trace. If those wear out, we can switch to solid copper.

GeorgesOatesLarsen, 2021-09-30:
Holy cow, ***that*** was just from ***hand spinning?!***

That's insane!

I thought we ordered 4oz from osh park? Or was it 2oz after all?

Either way, I kind of doubt that doubling the thickness will make much difference with osh, given what we've seen. Unless, perhaps, the gold plating is creating a softer alloy? I wonder if a run of unplated 4oz copper from Osh would fair better?

@petermccloud